gramaziokohler/integral_timber_joints #99: Planning `A34_M1` alone is okay but entire `seq_n 3` is stuck

yck011522, 2021-09-16:
**So If I call to plan only `A34_M1` like this, it will be successful:**
```
(itj3) PS C:\Users\leungp\Documents\GitHub\integral_timber_joints> p -m integral_timber_joints.planning.run --movement_id A34_M1 --load_external_movements --viewer --watch
pybullet build time: Aug  6 2021 15:05:25
Use Track IK: False
Arguments: Namespace(debug=False, design_dir='210605_ScrewdriverTestProcess', diagnosis=False, load_external_movements=True, low_res=False, max_distance=0.0, movement_id='A34_M1', problem='nine_pieces_process.json', problem_subdir='.', reinit_tool=False, seq_n=None, smooth=False, solve_mode='nonlinear', step_sim=False, verbose=True, viewer=True, watch=True, write=False)
==========
Process json parsed from c:\users\leungp\documents\github\integral_timber_joints\external\itj_design_study\210605_ScrewdriverTestProcess\nine_pieces_process.json
Loading external movements from c:\users\leungp\documents\github\integral_timber_joints\external\itj_design_study\210605_ScrewdriverTestProcess\results
Solving for movement A34_M1
--------------------
(3) Beam#b3
Beam b3 | Outer Trail movement_id #0
Process json parsed from c:\users\leungp\documents\github\integral_timber_joints\external\itj_design_study\210605_ScrewdriverTestProcess\nine_pieces_process.json
##########
Beam b3 | movement_id | Inner Trail #0 | time 0.00
Computing only for A34_M1
====================

* compute movement ids: ['A34_M1']
----------
(29)
RobotScrewdriverSyncLinearMovement(#A34_M1, Robot and Screwdriver (s2) syncronously move to retract from Beam ('b3'), traj 0)
WARNING: Attaching s2 (link gripper_base) to robot link robot11_tool0, but they are not in the same pose in pybullet scene.
-- gantry sampling iter 0
-- gantry sampling iter 1
-- gantry sampling iter 2
   -- IK iter 0
   -- IK iter 1
   -- IK iter 2
   -- IK iter 3
-- gantry sampling iter 3
-- gantry sampling iter 4
   -- IK iter 0
   -- IK iter 1
-- gantry sampling iter 5
-- gantry sampling iter 6
   -- IK iter 0
        cartesian trial #0
Plan found by IterativeIK! After 6 ik, 0 path failure over 7 samples.
~~~~~
        Propagate states for (29) : RobotScrewdriverSyncLinearMovement(#A34_M1, Robot and Screwdriver (s2) syncronously move to retract from Beam ('b3'), traj 1)
        - Altered (backward): (28) RoboticDigitalOutput(#A34_M0, Detach Screwdriver ('s2') from Beam ('b3'))
        - Altered (backward): (27) RoboticDigitalOutput(#A33_M3, Toolchanger Lock SL1 ('s2'))
        $ Impacted (backward): (26) RoboticLinearMovement(#A33_M2, Linear Advance to dock toolchanger of SL1 ('s2')., traj 0)
        $ Impacted (forward): (30) RoboticLinearMovement(#A34_M2, Linear retract after placing Beam ('b3'), traj 0)
====================

Visualize results
Start simulating results. Press enter to start.
WARNING: Attaching s2 (link gripper_base) to robot link robot11_tool0, but they are not in the same pose in pybullet scene.
===
Viz:
RobotScrewdriverSyncLinearMovement(#A34_M1, Robot and Screwdriver (s2) syncronously move to retract from Beam ('b3'), traj 1)
WARNING: Attaching s2 (link gripper_base) to robot link robot11_tool0, but they are not in the same pose in pybullet scene.
A plan has been found for beam id b3!
Return success: True
Planning done.
(itj3) PS C:\Users\leungp\Documents\GitHub\integral_timber_joints>
```

**However calling to run the entire `seq_n 3` will get stuck at A34_M1**

```
(itj3) PS C:\Users\leungp\Documents\GitHub\integral_timber_joints> p -m integral_timber_joints.planning.run --seq_n 3 --load_external_movements
pybullet build time: Aug  6 2021 15:05:25
Use Track IK: False
Arguments: Namespace(debug=False, design_dir='210605_ScrewdriverTestProcess', diagnosis=False, load_external_movements=True, low_res=False, max_distance=0.0, movement_id=None, problem='nine_pieces_process.json', problem_subdir='.', reinit_tool=False, seq_n=[3], smooth=False, solve_mode='nonlinear', step_sim=False, verbose=True, viewer=False, watch=False, write=False)
==========
Process json parsed from c:\users\leungp\documents\github\integral_timber_joints\external\itj_design_study\210605_ScrewdriverTestProcess\nine_pieces_process.json
Loading external movements from c:\users\leungp\documents\github\integral_timber_joints\external\itj_design_study\210605_ScrewdriverTestProcess\results
Solving for beam ['b3']
--------------------
(3) Beam#b3
Beam b3 | Outer Trail nonlinear #0
Process json parsed from c:\users\leungp\documents\github\integral_timber_joints\external\itj_design_study\210605_ScrewdriverTestProcess\nine_pieces_process.json
##########
Beam b3 | nonlinear | Inner Trail #0 | time 0.00
====================

* compute ['RoboticLinearMovement', 'RoboticClampSyncLinearMovement', 'RobotScrewdriverSyncLinearMovement'] (priority 1, status [<MovementStatus.neither_done: 5>, <MovementStatus.one_sided: 3>])
----------
(16)
RobotScrewdriverSyncLinearMovement(#A30_M2, Robot and Screwdrivers (['s1', 's2']) syncronously move to screw Beam ('b3'), traj 0)
WARNING: Attaching b3 (link BASE_LINK) to robot link robot11_tool0, but they are not in the same pose in pybullet scene.
WARNING: Attaching s1 (link gripper_base) to robot link robot11_tool0, but they are not in the same pose in pybullet scene.
WARNING: Attaching s2 (link gripper_base) to robot link robot11_tool0, but they are not in the same pose in pybullet scene.
WARNING: Attaching g2 (link gripper_base) to robot link robot11_tool0, but they are not in the same pose in pybullet scene.
-- gantry sampling iter 0
-- gantry sampling iter 1
   -- IK iter 0
   -- IK iter 1
   -- IK iter 2
   -- IK iter 3
-- gantry sampling iter 2
-- gantry sampling iter 3
   -- IK iter 0
   -- IK iter 1
   -- IK iter 2
   -- IK iter 3
-- gantry sampling iter 4
-- gantry sampling iter 5
-- gantry sampling iter 6
-- gantry sampling iter 7
   -- IK iter 0
   -- IK iter 1
-- gantry sampling iter 8
   -- IK iter 0
        cartesian trial #0
Plan found by IterativeIK! After 8 ik, 0 path failure over 9 samples.
~~~~~
        Propagate states for (16) : RobotScrewdriverSyncLinearMovement(#A30_M2, Robot and Screwdrivers (['s1', 's2']) syncronously move to screw Beam ('b3'), traj 1)
        $ Impacted (backward): (15) RoboticLinearMovement(#A30_M1, Linear Advance to bring Screwdriver tips to touch the predrilled hole., traj 0)
        - Altered (forward): (17) RoboticDigitalOutput(#A31_M0, Open Gripper ('g2') and let go of Beam ('b3'))
        $ Impacted (forward): (18) RoboticLinearMovement(#A31_M1, Linear retract after placing Beam ('b3'), traj 0)
----------
(29)
RobotScrewdriverSyncLinearMovement(#A34_M1, Robot and Screwdriver (s2) syncronously move to retract from Beam ('b3'), traj 0)
-- gantry sampling iter 0
   -- IK iter 0
   -- IK iter 1
   -- IK iter 2
   -- IK iter 3
-- gantry sampling iter 1
   -- IK iter 0
   -- IK iter 1
   -- IK iter 2
   -- IK iter 3
-- gantry sampling iter 2
-- gantry sampling iter 3
   -- IK iter 0
   -- IK iter 1
-- gantry sampling iter 4
   -- IK iter 0
   -- IK iter 1
-- gantry sampling iter 5
   -- IK iter 0
   -- IK iter 1
   -- IK iter 2
   -- IK iter 3
-- gantry sampling iter 6
   -- IK iter 0
   -- IK iter 1
-- gantry sampling iter 7
   -- IK iter 0
   -- IK iter 1
-- gantry sampling iter 8
   -- IK iter 0
   -- IK iter 1
```

yijiangh, 2021-09-23:
Not asking questions, but just posting pictures here so I can keep track of what's going on:

Taking a closer look, it seems that `s2` is at the wrong place, when planning sequentially:

[image]

yijiangh, 2021-09-23:
Ah I figured where the problem is, when planning sequentially, the previous movement and the current movement both have `s2` attached, but its grasp poses should be updated.

commit 3546560a (2021-09-24): Fix #99
closed 2021-09-24 by yijiangh