gramaziokohler/integral_timber_joints #110: Reorientation movement needs some fine-tuning

yijiangh, 2021-09-24:
After smoothing, the reorientation movement still looks like this:
[image]

I think here it's just that the end robot IK is sampled, and it's too far away from the start conf. We can either try turning this movement into a linear movement or be smarter about the sampling order.

yijiangh, 2021-09-24:
@yck011522 I'd recommend turning the reorientation movements into linear movements, and we can see if it works out-of-the-box or not. Opinion?

yck011522, 2021-09-24:
Worth giving it a try. @yijiangh 
Can do you a quick test form your side?

yijiangh, 2021-09-28:
My quick test shows that it's definitely worth a try:
https://user-images.githubusercontent.com/11905844/135121353-9334e49e-8c08-49df-a153-3c8a031ab910.mp4

yijiangh, 2021-09-28:
For 180-deg, it's possible but the z coordinate of the orient pose needs to be higher:

[image]
[image]

yijiangh, 2021-09-28:
This is `A135_M1` of hypar:

[image]

yijiangh, 2022-03-22:
Fixed by [237cf37](https://github.com/gramaziokohler/integral_timber_joints/commit/237cf379c3601e18cc03ebf461829df6fdd0585f)

closed 2022-03-22 by yijiangh