gramaziokohler/integral_timber_joints #112: check_ik breaks after 20 minutes

yck011522, 2021-09-27:
So I was trying to run check_ik on HyparHut, on this design repo commit: https://github.com/gramaziokohler/itj_design_study/commit/299a6d65ae36e88f738cc4ba77231ba99e1ba984

```p -m integral_timber_joints.planning.check_ik --design_dir 210419_AnticlasticShelter --problem shelter_process.json```

However after about 22 minutes. It ended with an error, see below. @yijiangh did you forgot to release some files after you open them. There should be no need to keep them open after each iteration.

```
Mov#774/State#775 (A243_M4) A(PickGripperFromStorageAction) M(RoboticLinearMovement) Linear Retract after getting PG1500 ('g3') from storage.
IK Success: IK found with pb-IK after 1/20 gantry attempts | total time 3.043
==========
Mov#775/State#776 (A244_M0) A(GenericGripperApproachBeamPickupAction) M(RoboticFreeMovement) Free Move to reach Pickup Approach Frame of Beam ('b19')
IK Success: IK found with pb-IK after 1/20 gantry attempts | total time 3.036
==========
Mov#777/State#778 (A244_M2) A(GenericGripperApproachBeamPickupAction) M(RoboticLinearMovement) Linear Advance to Storage Frame of Beam ('b19')
IK Success: IK found with pb-IK after 1/20 gantry attempts | total time 3.026
==========
Mov#780/State#781 (A247_M0) A(PickAndRotateBeamForAttachingScrewdriverAction) M(RoboticLinearMovement) Linear Move to lift up Beam ('b19')
IK Success: IK found with pb-IK after 1/20 gantry attempts | total time 3.017
==========
Mov#781/State#782 (A247_M1) A(PickAndRotateBeamForAttachingScrewdriverAction) M(RoboticFreeMovement) Free Move to reorient Beam ('b19')
IK Success: IK found with pb-IK after 1/20 gantry attempts | total time 2.945
==========
Mov#784/State#785 (A250_M0) A(AssembleBeamWithScrewdriversAction) M(RoboticFreeMovement) Free Move to bring Beam ('b19') to assemble_approach position on structure.
IK Success: IK found with IKFast after 1/20 gantry attempts - 0 ik attempts / total 6 IKFast solutions | total time 2.988
==========
Mov#785/State#786 (A250_M1) A(AssembleBeamWithScrewdriversAction) M(RoboticLinearMovement) Linear Advance to bring Screwdriver tips to touch the predrilled hole.
IK Success: IK found with pb-IK after 1/20 gantry attempts | total time 3.129
==========
Mov#786/State#787 (A250_M2) A(AssembleBeamWithScrewdriversAction) M(RobotScrewdriverSyncLinearMovement) Robot and Screwdrivers (['s1', 's2']) syncronously move to screw Beam ('b19')
Traceback (most recent call last):
  File "C:\Users\leungp\.conda\envs\itj3\lib\runpy.py", line 193, in _run_module_as_main
  File "C:\Users\leungp\.conda\envs\itj3\lib\runpy.py", line 85, in _run_code
  File "c:\users\leungp\documents\github\integral_timber_joints\src\integral_timber_joints\planning\check_ik.py", line 82, in <module>
  File "c:\users\leungp\documents\github\integral_timber_joints\src\integral_timber_joints\planning\rhino_interface.py", line 35, in get_ik_solutions
  File "c:\users\leungp\documents\github\integral_timber_joints\src\integral_timber_joints\planning\state.py", line 149, in set_state
  File "c:\users\leungp\documents\github\integral_timber_joints\external\compas_fab_pychoreo\external\pybullet_planning\src\pybullet_planning\interfaces\env_manager\user_io.py", line 57, in __enter__
OSError: [Errno 24] Too many open files
(itj3) PS C:\Users\leungp\Documents\GitHub\integral_timber_joints>
```

yck011522, 2021-09-27:
Hum.. And there is no easy way to restart the check_ik from a specific id....

yck011522, 2021-09-27:
I added a PowerShell Script to continue using it for now.

yijiangh, 2021-09-28:
OK I fixed both the OSError and provided the `--seq_n` option for multi-beam planning which works in the same way as `run.py`.

I also added an option to pass in `client` and `robot` so when running from the script without rpc proxy, it will reuse the client and save a LOT of time. I run `p -m integral_timber_joints.planning.check_ik --design_dir 210419_AnticlasticShelter --problem shelter_process.json` to check all the beams and it finishes under 9 mins.

```
Failed State Indices: [53, 1469, 2428, 2566]
Total checking time: 512.443
```

yijiangh, 2021-09-28:
Turns out all of the four movements above have IK solutions if more gantry attempts are given. Now, we support both str-based movement id and index-based movement id. So you can do both:
```
p -m integral_timber_joints.planning.check_ik --design_dir 210419_AnticlasticShelter --problem shelter_process.json --movement_id A805_M0
``` 
or 
```
p -m integral_timber_joints.planning.check_ik --design_dir 210419_AnticlasticShelter --problem shelter_process.json --movement_id 2565
```
Don't forget that the failed movement index = state index -1.

yijiangh, 2021-09-28:
Also, you can use the `-v --debug` flag if you want to see the collision diagnosis info visually:

I am amazed by the scale of our hypar hut setup 😺 
[image]

closed 2021-09-28 by yijiangh

yck011522, 2021-09-28:
This is an amazing fix. Thanks a lot.

commit 8c27aa0c (2021-09-28): Fix #112