gramaziokohler/integral_timber_joints #120: HyparHut `b26` stuck in one attempt for hours.

yck011522, 2021-10-05:
I was rrunning a full planning routine for HyparHut Round 2 with the following command ```p -m integral_timber_joints.planning.run --design_dir 210419_AnticlasticShelter --problem shelter_process.json --reinit_tool --write --load_external_movements --solve_timeout 1200 --smooth``` 

It was all okay until `b26` where it got stuck for 4 hours in the big for loop of trying the linear buffer. This should not be what the buffer routine intended to do because we expect it to be a strategy to deal with narrow paths, However, I think in most of these attempts, the linear motion was actually successful, yet the free motion is not. Perhaps this implies it is not a narrow path problem.

I'll need to further investigate the reason why this element is so difficult.

```
----------
(1134)
RoboticFreeMovement(#A362_M0, Free Move to bring Beam ('b26') to assemble_approach position on structure., traj 0)
Free motion | START linear buffer: trying retraction dist 0.0000 in [1. 0. 0.]
Start 0.0 retraction dist along [1. 0. 0.]
- Free motion | END linear buffer: trying retraction dist 0.0000 in [1. 0. 0.]
End 0.0 retraction dist along [1. 0. 0.]
No free motion found with algorithm birrt!
No free motion found under current retraction (st [1. 0. 0.]|0.0000 ; end [1. 0. 0.]|0.0000).
- Free motion | END linear buffer: trying retraction dist 0.0500 in [1. 0. 0.]
End cart traj found.
No free motion found with algorithm birrt!
No free motion found under current retraction (st [1. 0. 0.]|0.0000 ; end [1. 0. 0.]|0.0500).
- Free motion | END linear buffer: trying retraction dist 0.1000 in [1. 0. 0.]
No Cartesian motion found, due to IK plan is not found.!
No end cart traj found.
- Free motion | END linear buffer: trying retraction dist 0.0000 in [0. 1. 0.]
End 0.0 retraction dist along [0. 1. 0.]
No free motion found with algorithm birrt!
No free motion found under current retraction (st [1. 0. 0.]|0.0000 ; end [0. 1. 0.]|0.0000).
- Free motion | END linear buffer: trying retraction dist 0.0500 in [0. 1. 0.]
End cart traj found.
No free motion found with algorithm birrt!
No free motion found under current retraction (st [1. 0. 0.]|0.0000 ; end [0. 1. 0.]|0.0500).
- Free motion | END linear buffer: trying retraction dist 0.1000 in [0. 1. 0.]
End cart traj found.
No free motion found with algorithm birrt!
No free motion found under current retraction (st [1. 0. 0.]|0.0000 ; end [0. 1. 0.]|0.1000).
- Free motion | END linear buffer: trying retraction dist 0.0000 in [0. 0. 1.]
End 0.0 retraction dist along [0. 0. 1.]
No free motion found with algorithm birrt!
No free motion found under current retraction (st [1. 0. 0.]|0.0000 ; end [0. 0. 1.]|0.0000).
- Free motion | END linear buffer: trying retraction dist 0.0500 in [0. 0. 1.]
No Cartesian motion found, due to IK plan is not found.!
No end cart traj found.
- Free motion | END linear buffer: trying retraction dist 0.1000 in [0. 0. 1.]
No Cartesian motion found, due to IK plan is not found.!
No end cart traj found.
- Free motion | END linear buffer: trying retraction dist 0.0000 in [-1. -0. -0.]
End 0.0 retraction dist along [-1. -0. -0.]
No free motion found with algorithm birrt!
No free motion found under current retraction (st [1. 0. 0.]|0.0000 ; end [-1. -0. -0.]|0.0000).
- Free motion | END linear buffer: trying retraction dist 0.0500 in [-1. -0. -0.]
End cart traj found.
No free motion found with algorithm birrt!
No free motion found under current retraction (st [1. 0. 0.]|0.0000 ; end [-1. -0. -0.]|0.0500).
- Free motion | END linear buffer: trying retraction dist 0.1000 in [-1. -0. -0.]
End cart traj found.
No free motion found with algorithm birrt!
No free motion found under current retraction (st [1. 0. 0.]|0.0000 ; end [-1. -0. -0.]|0.1000).
- Free motion | END linear buffer: trying retraction dist 0.0000 in [-0. -1. -0.]
End 0.0 retraction dist along [-0. -1. -0.]
No free motion found with algorithm birrt!
No free motion found under current retraction (st [1. 0. 0.]|0.0000 ; end [-0. -1. -0.]|0.0000).
- Free motion | END linear buffer: trying retraction dist 0.0500 in [-0. -1. -0.]
End cart traj found.
No free motion found with algorithm birrt!
No free motion found under current retraction (st [1. 0. 0.]|0.0000 ; end [-0. -1. -0.]|0.0500).
- Free motion | END linear buffer: trying retraction dist 0.1000 in [-0. -1. -0.]
End cart traj found.
No free motion found with algorithm birrt!
No free motion found under current retraction (st [1. 0. 0.]|0.0000 ; end [-0. -1. -0.]|0.1000).
- Free motion | END linear buffer: trying retraction dist 0.0000 in [-0. -0. -1.]
[... 529 lines cut ...]
- Free motion | END linear buffer: trying retraction dist 0.1000 in [0. 1. 0.]
End cart traj found.
No free motion found with algorithm birrt!
No free motion found under current retraction (st [0. 0. 1.]|0.1000 ; end [0. 1. 0.]|0.1000).
- Free motion | END linear buffer: trying retraction dist 0.0000 in [0. 0. 1.]
End 0.0 retraction dist along [0. 0. 1.]
No free motion found with algorithm birrt!
No free motion found under current retraction (st [0. 0. 1.]|0.1000 ; end [0. 0. 1.]|0.0000).
- Free motion | END linear buffer: trying retraction dist 0.0500 in [0. 0. 1.]
No Cartesian motion found, due to IK plan is not found.!
No end cart traj found.
- Free motion | END linear buffer: trying retraction dist 0.1000 in [0. 0. 1.]
No Cartesian motion found, due to IK plan is not found.!
No end cart traj found.
- Free motion | END linear buffer: trying retraction dist 0.0000 in [-1. -0. -0.]
End 0.0 retraction dist along [-1. -0. -0.]
No free motion found with algorithm birrt!
No free motion found under current retraction (st [0. 0. 1.]|0.1000 ; end [-1. -0. -0.]|0.0000).
- Free motion | END linear buffer: trying retraction dist 0.0500 in [-1. -0. -0.]
End cart traj found.
No free motion found with algorithm birrt!
No free motion found under current retraction (st [0. 0. 1.]|0.1000 ; end [-1. -0. -0.]|0.0500).
- Free motion | END linear buffer: trying retraction dist 0.1000 in [-1. -0. -0.]
End cart traj found.
```

yck011522, 2021-10-05:
@yijiangh I think I need a way to **turn off** or **limit the steps of** the Free Motion Start / End buffer.

yijiangh, 2021-10-05:
Yes, there are even duplicated linear buffer motion trials if you look closely. I will try to fix this. Related to #117 .

The testing movement will be `A362_M0`.

yck011522, 2021-10-06:
Hum. I did not spot the duplicate. please double check.

yijiangh, 2021-10-18:
Hmm, I am puzzled. This movement shouldn't be that hard to plan: